TheKikGen/USBMidiKliK #1: Arduino Uno not detected on USB bus after flashing

DLehenbauer, 2018-05-12:
Successfully flashed the 16u2 on an Arduino Uno R3 clone (Elegoo), but after disconnecting the ISP programmer and unplugging/reconnecting the USB cable to the Uno, the device was not detected on the USB bus.

It wasn't just unrecognized or reported as malfunctioning, Windows 10 didn't detect that a new USB device had been plugged in.  (This happened both with and without a jumper across ISP pins 4/6.)

**Note:** The same steps using 'mocolufa' works for me, leading me to believe it's a bug or incompatibility with USBMidiKliK as opposed to user error.)

```
> .\avrdude.exe -c usbtiny -P usb -p m16u2 -b 19200 -U flash:w:USBMidiKliK_dual.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude.exe: Device signature = 0x1e9489 (probably m16u2)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "USBMidiKliK_dual.hex"
avrdude.exe: writing flash (8644 bytes):

Writing | ################################################## | 100% 10.57s

avrdude.exe: 8644 bytes of flash written
avrdude.exe: verifying flash memory against USBMidiKliK_dual.hex:
avrdude.exe: load data flash data from input file USBMidiKliK_dual.hex:
avrdude.exe: input file USBMidiKliK_dual.hex contains 8644 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 7.27s

avrdude.exe: verifying ...
avrdude.exe: 8644 bytes of flash verified

avrdude.exe: safemode: Fuses OK (E:F4, H:D9, L:FF)

avrdude.exe done.  Thank you.
```

TheKikGen, 2018-05-18:
Did you compiled the code or have you uploaded .hex file provided in the Git ?

DLehenbauer, 2018-05-19:
Yep, that fixed it.  Works great now.  Thx!

closed 2018-05-19 by DLehenbauer