TheKikGen/USBMidiKliK #3: Missing Bytes on SysEx

mungewell, 2018-07-02:
For some reason my code sending SysEx is missing bytes, this seems to happen when sending 6 bytes, but not 7
```
    stream.write( 0xF0 );
    stream.write( 0x7F );
    stream.write( 0x7F );
    stream.write( 0x06 );
    stream.write( 0x02 );
    stream.write( 0xF7 );
522653696: 0xf07f7ff7
    stream.write( 0xF0 );
    stream.write( 0x7F );
    stream.write( 0x7F );
    stream.write( 0x06 );
    stream.write( 0x02 );
    stream.write( 0x02 );
    stream.write( 0xF7 );
527349760: 0xf07f7f060202f7
```

mungewell, 2018-07-04:
Monitoring the UART link between ATMegas on the board with BusPirate (31250baud = BRG 127) I can see that "f0 7f 7f 06 01 f7" is sent, but comes out of USB/Midi interface as "f0 7f 7f f7 01 00".

```
$ amidi -p hw:1,0,0 -r ~/temp.bin
^C
21 bytes read
$ hexdump -Cv !$
hexdump -Cv ~/temp.bin
00000000  f0 7f 7f f0 7f 7f f7 01  00 f0 7f 7f f7 01 00 f0  |................|
00000010  7f 7f f7 01 00                                    |.....|
00000015
$ hexdump -Cv ~/minicom.cap 
00000000  28 32 29 0a 52 61 77 20  55 41 52 54 20 69 6e 70  |(2).Raw UART inp|
00000010  75 74 0a 41 6e 79 20 6b  65 79 20 74 6f 20 65 78  |ut.Any key to ex|
00000020  69 74 0a f0 7f 7f 06 01  f7 f0 7f 7f 06 01 f7 f0  |it..............|
00000030  7f 7f 06 01 f7 f0 7f 7f  06 01 f7 0a 55 41 52 54  |............UART|
00000040  3e                                                |>|
00000041
```

TheKikGen, 2018-08-31:
HI. Thanks for that.  I'l will have a look rapidly.

VZtec, 2018-09-26:
Hello, congratulations for the project.

I had the same problem of losing sysex messages from the arduino to the PC.

I found the error on line 469 of the "USBMidiKliK_dual.cpp" source file:

`MIDIEvent.Data1 = midiXparser::eoxStatus;`

 EOX byte is always set in data1 when it can be any of the 3 datas, depending on the length of the message.

My solutions was to replace line 469 with:

`if(serialMidiParser->isSysExError())`
`MIDIEvent.Data1 = midiXparser::eoxStatus;`
`else`
`((uint8_t*)&MIDIEvent)[packetLen] = readByte;`

That solves the problem :)

Thank you very much for the code.

Greetings from Argentina.

Franco

TheKikGen, 2018-10-01:
At fist, I was thinking the issue was coming from my midi parser object libray, but it passed all the automated tests, and hopefully as I use it in a lot of other projects.

For sure the issue was in the parent code....you find it !
Very good point Franco !

TheKikGen, 2018-10-01:
Patch to the master branch.  Binary not still updated.

TheKikGen, 2018-10-01:
Hex file uploaded to the repository. Build 649. Issue closed.

closed 2018-10-01 by TheKikGen