TheKikGen/USBMidiKliK #5: midi Data[0] incorrect when send Program Change

margat78, 2018-12-06:
when i try to send a PC message it sends  last sent Data byte in midi data[0].
Step to reproduce:
1) send midi CC 34 value 44 on channel 5 **OK**
2) try to send  ProgramChange 12 on channel 5 -> it sends midi PC 44 on channel 5

with hiduno works. (hiduno has problem with PC inputs instead USBMidiKliK works great!)

TheKikGen, 2018-12-08:
Hello. Thanks ! 

It is a stupid bug in the midiXparser.  I was forcing 3 as the length for all channel msg....don't know why :-) ! Tested and works.  Please confirm with the last bin compiled at master release.  
If not, reopen the issue.

MIDI-OX results :

 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
 00002B9B   5  --     B0    00    00    1  ---  CC: Bank MSB          
 00002B9B   5  --     B0    20    00    1  ---  CC: Bank LSB          
 00002B9C   5  --     C0    1C    --    1  ---  PC: Electric Guitar (muted) 
 00009207   5  --     90    3A    64    1  Bb 3 Note On               
 0000924F   5  --     80    3A    40    1  Bb 3 Note Off              
 0000BE3B   5  --     B0    00    00    1  ---  CC: Bank MSB          
 0000BE3B   5  --     B0    20    00    1  ---  CC: Bank LSB          
 0000BE3C   5  --     C0    16    --    1  ---  PC: Harmonica         
 00010F17   5  --     B0    01    52    1  ---  CC: Modulation        
 000113C7   5  --     B0    01    53    1  ---  CC: Modulation        
 00013A33   5  --     B0    00    00   1  ---  CC: Bank MSB          
 00013A33   5  --     B0    20    00    1  ---  CC: Bank LSB          
 00013A34   5  --     C0    0A    --    1  ---  PC: Music Box

closed 2018-12-08 by TheKikGen

margat78, 2018-12-09:
it works perfectly!
tks a lot