TheKikGen/USBMidiKliK #6: build instructions

EsGeh, 2019-05-12:
Would be nice to have a few lines how to build the bootloader, for unexperienced users like me.
(naively typing `make` leads to an error because of hardcoded windows (!) path...)

orzdk, 2019-05-28:
I just got this to build. What exact error do you get? If you mean the AVRDUDE_PATH I guess that should not be an issue until actually uploading, seems to make ok with a wrong path. 

Things i needed to fix was to be sure to use LUFA github master, there seems to be changes to the interfaces since 170418. Be sure to put lufa top level folder, not LUFA subfolder, in same folder as the USBMidiKlik folder.

Then download the midiXParser (also from this user) and put the folder in same folder level as USBMidiklik and lufa folders, then references will line up. Then i had to change a prototype, line 117 in midiXParser.h should be 

void        setMidiMsgFilter(uint8_t midiMsgTypeFilterMsk);

After those small issues it seems to make without problems, of course assuming you have a working avr-gcc environment. Will check upload tomorrow, but that should probably also work fine if you point the AVRDUDE_PATH to the arduino ide bin folder.

HobbyEE, 2019-05-28:
> I just got this to build. What exact error do you get? If you mean the AVRDUDE_PATH I guess that should not be an issue until actually uploading, seems to make ok with a wrong path.
>
> Things i needed [...]

That would be great! I am also trying to make it on Mac Os X Mojave but I keep getting no target found. Any tip would be great!

orzdk, 2019-05-29:
It seems like it just cannot find the lufa library folder. The path is relative to the location of the makefile, make sure to structure it like: 

--workfolder
----lufa
----midiXParser
----USBMidiKlik

HobbyEE, 2019-05-29:
Thank you very much! I was nesting lufa and midiXparser inside USBMidiKliK folder... Below is the output of **make** command. I get some warnings but I also got a hex file, which is good... Now I have to remember how to use USBtinyISP to burn it on my Mega (all my UNOs have CH340).

`# Create an auto-incrementing build number.
======================================================================
======== BUILD USBMIDIKLIK - TARGET=uno - BUILD NO=1085 ========
======================================================================
 [INFO]    : Begin compilation of project "USBMidiKliK_dual_uno"...

avr-gcc (GCC) 4.8.1
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

 [GCC]     : Compiling C++ file "USBMidiKliK_dual.cpp"
avr-gcc -c -pipe -gdwarf-2 -g2 -mmcu=atmega16u2 -fshort-enums -fno-inline-small-functions -Wall -fno-strict-aliasing -funsigned-char -funsigned-bitfields -ffunction-sections -I. -DARCH=ARCH_AVR8 -DF_CPU=16000000UL -mrelax -fno-jump-tables -x c++ -Os -std=gnu++98 -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -DARDUINO_DEVICE_VENDORID=0x2341 -DARDUINO_DEVICE_PRODUCTID=0x0001 -DARDUINO_DEVICE_MANUFACTURER_STRING="Arduino (www.arduino.cc)" -DARDUINO_DEVICE_PRODUCT_STRING="Arduino Uno dual midi" -DARDUINO_DEVICE_PRODUCT_SERIAL="55732323430351718180" -DMIDI_DEVICE_VENDORID=0x2912 -DMIDI_DEVICE_PRODUCTID=0x1967 -DMIDI_DEVICE_MANUFACTURER_STRING="The KikGen Labs" -DMIDI_DEVICE_PRODUCT_STRING="USB MidiKliK Build-1086                      "  -DBUILD_NUMBER=1086 -DBUILD_DATE=2019.05.29-14:26:23 -DUSE_LUFA_CONFIG_HEADER -IConfig/ -fpermissive -Os  -I. -I../lufa/LUFA/.. -DARCH=ARCH_AVR8 -DBOARD=BOARD_UNO -DF_USB=16000000UL   -MMD -MP -MF obj/USBMidiKliK_dual.d USBMidiKliK_dual.cpp -o obj/USBMidiKliK_dual.o
USBMidiKliK_dual.cpp:171:3: warning: #warning "DUAL BOOT MODE - JUMPER & SOFT" [-Wcpp]
  #warning "DUAL BOOT MODE - JUMPER & SOFT"
   ^
In file included from USBMidiKliK_dual.cpp:21:0:
USBMidiKliK_dual.h: In function 'void ConfigRootMenu()':
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
  static void USBSerialPutStr(char *,bool);
              ^
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.h:84:14: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.cpp: In function 'void ShowCurrentSettings()':
USBMidiKliK_dual.cpp:365:13: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
 static void USBSerialPutStr(char *s, bool pgm) {
             ^
USBMidiKliK_dual.cpp:365:13: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.cpp:365:13: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.cpp:365:13: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
USBMidiKliK_dual.cpp:365:13: warning:   initializing argument 1 of 'void USBSerialPutStr(char*, bool)' [-fpermissive]
[... cut ...]
 [NM]      : Extracting SYM file data from "USBMidiKliK_dual_uno.elf"
avr-nm -n USBMidiKliK_dual_uno.elf > USBMidiKliK_dual_uno.sym
 [SIZE]    : Determining size of "USBMidiKliK_dual_uno.elf"

avr-size --mcu=atmega16u2 --format=avr USBMidiKliK_dual_uno.elf
AVR Memory Usage
----------------
Device: atmega16u2

Program:   10098 bytes (61.6% Full)
(.text + .data + .bootloader)

Data:        371 bytes (72.5% Full)
(.data + .bss + .noinit)

 [INFO]    : Finished building project "USBMidiKliK_dual_uno".
`

HobbyEE, 2019-05-29:
OK, I gave 
`avrdude -p m16u2 -F -P usb -c usbtiny -U flash:w:USBMidiKliK_dual_uno.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m`
on the terminal. Everything went fine and now my Arduino Mega appears as "USB MidiKliK Build-1083" on my Mac! So far so good! Now I need to get some examples on my hand to get something useful out of this...

orzdk, 2019-05-29:
Nice that so freshly aquired knowledge by me was still able to help you, and you are most welcome. Basically I am also a novice in the AVR world, so that was luck i guess : )

I believe that I also got those warnings, but it finished, so I am also looking forward to try to upload later today, - in my case to a Pro Micro ATmega32u4. 

PM me so we can keep in contact, if we're working on the same things we might as well knowledge share. 

VBR

TheKikGen, 2019-06-06:
Hi Guys.  Nice to see mutual support between users.   If any of you feel ready to write a wiki on how to build USBMidiKlik (which I don't really have time to do ) he is welcome!
Know that some specific options in the makefile allows you to build also the version for Arduino Leonardo/Nano.

closed 2019-06-06 by TheKikGen

EsGeh, 2019-06-28:
Thanks for all the information.
I have been able to compile, after some tinkering.
I had to solve compile errors by figuring out version conflicts between the libraries involved, by hand (and wild guessing).
I still think it would be nice to have some information about the dependencies (including version/git commits) in the README. This only makes sense if the developers keep them up to date, of course.
In my own projects I handle this by having (bash-/fish-/...) scripts to automatize downloading dependencies, building, and a "config" file which specifies the git hashes of the dependencies needed.
I can offer to provide similar scripts for this project, if wished.