TheKikGen/USBMidiKliK #8: Arduino Mega?

EsGeh, 2019-06-30:
Hey!

How can I use the firmware with a Arduino Mega?
So far I have been using the version for Uno.
This works, somehow.
But when I turn the device into USB serial mode, the device is recognized as an Arduino Uno, which leads to difficulties.

Is there a simple solution?

TheKikGen, 2019-07-02:
# Specify the Vendor ID, Product ID and device name ,in the makefile.
# This is used by Descriptors.c
ARDUINO_DEVICE_VENDORID							= 0x2912
ARDUINO_DEVICE_PRODUCTID 						= 0x0001
ARDUINO_DEVICE_MANUFACTURER_STRING 	= "The KikGen Labs"
ARDUINO_DEVICE_PRODUCT_STRING 			= "Arduino Micro dual midi"
ARDUINO_DEVICE_PRODUCT_SERIAL 			= "55732323430351718180"

closed 2019-07-07 by TheKikGen

EsGeh, 2019-07-07:
Thanks a lot, it seems to work

luca-art, 2020-07-01:
Hi and thank you for your code.
I made a Jupiter-8 (Arturia plugin) controller over a MEGA. Everything works fine when I tried your software over an UNO (reset Arduino, Atmel Flip, etc) but now, i have to inject it over the MEGA and I do not know how can i do it. I do not have hex file for MEGA and I do notknow how compilate it. I read your prev post and I spent half of the morning figuring out the way to do it, but I am not capable. Is it a process involving the command line? I am a newby in programing. Thank you

luca-art, 2020-07-03:
I am sorry I saw you closed this post