TheKikGen/USBMidiKliK #15: Go into "dual mode" using an Arduino Mega (ATmega16U2)

khannurien, 2021-01-03:
Hi,

First, thank you for your hard work. The firmware works perfectly for me -- I managed to build it for the Mega using [EsGeh's pull request](https://github.com/TheKikGen/USBMidiKliK/pull/7).

I'd like to use the dual mode advertised in the README:

> For more convenience when updates are needed, a "dual mode" is embedded, allowing to switch back to the USB serial : when the PB2/MOSI pin of the ATMEGA8U2 is connected to ground, the Arduino is a classical one again, and you can change and upload a new firmware in the ATMEGA328P (UNO) with the standard Arduino IDE.

However, I can't seem to achieve that. I tried both the following bridging (reference in links): [GND to Digital 51](http://sin.lyceeleyguescouffignal.fr/wp-content/uploads/2018/10/fed21bas-960.jpg), [GDN to ICSP PB2](https://www.circuitstoday.com/wp-content/uploads/2018/02/Arduino-Mega-ICSP-Pinout.png). None of them produced the expected result.

Am I missing something?

Thanks!

TheKikGen, 2021-01-20:
Hi,
The principle of the dual mode is to read a pin at boot time, and if grounded, then to change the behaviour of the bootloader.

1/ Your Arduino must have an ATMEGA8U2 or a ATMEGA 16U2 managing the usb (a small square microcontroler next to the USB socket).  Some Mega clones use CH340...

2/ You must ground the PB2/MOSI pin of the ATMEGA8U2/ATMEGA 16U2 (not the ATmega2560 uC)
You will find that pin on the 6 Pins connector labelled "ICSP"
https://www.circuitstoday.com/arduino-mega-pinout-schematics

khannurien, 2021-01-20:
I believe I did not reboot the Arduino after grounding the pin and that is why it didn't work. I'll come back and close the issue in a few days when I have time to test it again.

Thank you!

closed 2023-10-27 by TheKikGen