TheKikGen/USBMidiKliK #16: Error: unrecognized command line option

bambax, 2021-01-12:
Hi,

When trying to make I get the error:

`cc1plus.exe: error: unrecognized command line option "-std=gnu++11"`

Environment is Windows 7 64bits with WinAVR 2010-01-20 (latest available version from SourceForge).

Is this a problem with the gcc version of WinAVR? How to troubleshoot this issue?

Thanks!

tttapa, 2021-01-12:
A compiler version from 2010 won't have support for C++11 (released in 2011). You'll have to find a newer compiler.

bambax, 2021-01-12:
Ok but the compiler recommended by the wiki for this project is this one... AFAIK there is no version of WinAVR more recent than this one.

The actual file is called `cc1plus.exe`. I have tried to replace it with a more recent version distributed by Arduino but then it complains that the architecture is not supported.

I have also tried to change the flags. In the makefile there is this:

    # From gcc 4.7
    CC_FLAGS     = -std=gnu++11
    # Before gcc 4.7
    #CC_FLAGS     += -std=c++0x

but the std option is also set in gcc.mk; if we change it there too, compiling start but there are a lot of errors and warnings (most common one being `only initialized variables can be placed into program memory area`) and it eventually fails.

Has this been made to work on Windows...? Should one find a specific version of  `cc1plus.exe`?

tttapa, 2021-01-12:
I've never used WinAVR, but reading the reviews, it looks like terrible software: https://sourceforge.net/projects/winavr/reviews/?stars=0#reviews

I can only recommend Linux for this kind of development.

Never replace a single executable like that, it's part of the toolchain, you can't just mix and match components from different versions.
You could try using the Arduino distribution of avr-gcc, but not in combination with WinAVR.

TheKikGen, 2021-01-20:
Hi.  
Try these ones :
https://www.microchip.com/en-us/development-tools-tools-and-software/gcc-compilers-avr-and-arm
https://blog.zakkemble.net/avr-gcc-builds/

TheKikGen, 2021-01-20:
other sources :
https://sourceforge.net/projects/mobilechessboar/files/avr-gcc%20snapshots%20%28Win32%29/

bambax, 2021-01-21:
Hi,

Many thanks for answering!
I did eventually succeed in compiliing by doing ungodly things (copying and mixing files from different sources).
And then I realized I didn't need to, since a hex file is already available for each platform... ;-)
Maybe you can add a line at the top of the readme file to remind people that they can simply use the provided compiled files?
Anyway this is most useful, many thanks!!

closed 2023-12-21 by TheKikGen