TheKikGen/USBMidiKliK #17: HEX file

Lucmunoz, 2021-06-13:
Hello! Best regards from Chile. 

Well this is not an issue at all... I would like to ask for help. Is there a way that you could upload a build hex file? I Don't know how to compile it.  

I know it has to be made using WINAVR or withina avr-gcc environment that I don't know how to execute since I usually only work with arduino IDE.

Hope you can helpme or maybe explain me how to do it. Thanks so much Sr.

TheKikGen, 2023-12-20:
If you have a Linux OS, check that :  https://github.com/TheKikGen/USBMidiKliK/wiki
If not, you can use a virtual machine.

Under Windows, use avrdude : 

`avrdude -v -patmega32u4 -cavr109 -PCOM5 -b57600 -D -V -Uflash:w:./USBMidiKliK_dual.hex:i`

TheKikGen, 2023-12-23:
Mega hex file added to the repo